addiejanssen/domoticz-solaredge-modbustcp-plugin #4: Power (kWh)

Frank18000, 2020-10-11:
In domoticz i see an  measurement of 451215.360 kWh. I think there is a decimal on the wrong place. Can i correct that my self?
--

addiejanssen, 2020-10-12:
Hi @Frank18000,

I assume that you see this in the Setup => Devices list. Is that correct?
I struggled with this for a while, since all other plugins that use this device type also store the Wh value, not the kWh value.
When looking at the graph in the Utility tab, it has the correct scale for my setup.
It is only showing a weird value in the Setup => Devices list.

Can you check your graph (if possible, take a screenshot of both the device list and the graph and upload it here)?

Frank18000, 2020-10-12:
This is wat is see in de device list and the graphs

[image]
[image]

addiejanssen, 2020-10-12:
Interesting...

Do you have another device (for example P1 - smart meter) setup that is reporting power in kWh?
Do you use the SolarEdge API to collect information from your inverter?
I wonder what they would show.

Would you be so kind to export the "Week" to a csv file and then show the csv file from a text editor.
So don't open it in something like Excel; we don't want to have an application format the values.
What I hope to see is something like:

"DateTime","Energy Generated"
"2020-10-05 00:00:00",5.605
"2020-10-06 00:00:00",4.315
"2020-10-07 00:00:00",6.818
"2020-10-08 00:00:00",2.125
"2020-10-09 00:00:00",6.694
"2020-10-10 00:00:00",8.729
"2020-10-11 00:00:00",8.879
"2020-10-12 00:00:00",4.769

Frank18000, 2020-10-12:
i don't have an api for my solar edge just your system that works great except for this little thing. I have an P1 meter that is reporting inkWh

this is fom the solar edge
"DateTime";"Energie opgewekt";"Vermogen gegenereerd"
"2020-10-11 15:00:00";57160500;567
"2020-10-11 15:05:00";;465
"2020-10-11 15:10:00";;580,6
"2020-10-11 15:15:00";;754,4
"2020-10-11 15:20:00";;828
"2020-10-11 15:25:00";;919,8
"2020-10-11 15:30:00";;982,4
"2020-10-11 15:35:00";;1272,8
"2020-10-11 15:40:00";;1026,4
"2020-10-11 15:45:00";;820,8
"2020-10-11 15:50:00";;900,2
"2020-10-11 15:55:00";;914,8
"2020-10-11 16:00:00";65182104;830,6
"2020-10-11 16:05:00";;490,8
"2020-10-11 16:10:00";;1207,6
"2020-10-11 16:15:00";;1932
"2020-10-11 16:20:00";;1212,8
"2020-10-11 16:25:00";;944
"2020-10-11 16:30:00";;810,2
"2020-10-11 16:35:00";;961,4
"2020-10-11 16:40:00";;1006,2
"2020-10-11 16:45:00";;749,4
"2020-10-11 16:50:00";;896
"2020-10-11 16:55:00";;351,2
"2020-10-11 17:00:00";20984628;360
"2020-10-11 17:05:00";;418,2
"2020-10-11 17:10:00";;422,6
"2020-10-11 17:15:00";;738,6
"2020-10-11 17:20:00";;416,4
"2020-10-11 17:25:00";;278,4
"2020-10-11 17:30:00";;268,6
"2020-10-11 17:35:00";;243,2
"2020-10-11 17:40:00";;225,4
"2020-10-11 17:45:00";;186
"2020-10-11 17:50:00";;143
"2020-10-11 17:55:00";;140,6
"2020-10-11 18:00:00";1350042;150,4
"2020-10-11 18:05:00";;73,4
"2020-10-11 18:10:00";;20,4
"2020-10-11 18:15:00";;10,8
"2020-10-11 18:20:00";;16,6
"2020-10-11 18:25:00";;10,6
"2020-10-11 18:30:00";;22,8
"2020-10-11 18:35:00";;33,2
"2020-10-11 18:40:00";;13,8
"2020-10-11 18:45:00";;11,6
"2020-10-11 18:50:00";;1
"2020-10-11 18:55:00";;0
"2020-10-11 19:00:00";0;0
"2020-10-11 19:05:00";;0
"2020-10-11 19:10:00";;0
"2020-10-11 19:15:00";;0
"2020-10-11 19:20:00";;0
"2020-10-11 19:25:00";;0
"2020-10-11 19:30:00";;0
"2020-10-11 19:35:00";;0
"2020-10-11 19:40:00";;0
"2020-10-11 19:45:00";;0
"2020-10-11 19:50:00";;0
"2020-10-12 08:00:00";1795686
"2020-10-12 08:20:00";;9,4
"2020-10-12 08:25:00";;12,6
"2020-10-12 08:30:00";;11,6
"2020-10-12 08:35:00";;21
"2020-10-12 08:40:00";;35,4
"2020-10-12 08:45:00";;66,4
"2020-10-12 08:50:00";;80
"2020-10-12 08:55:00";;86,4
"2020-10-12 09:00:00";13434880;92
"2020-10-12 09:05:00";;103,4
"2020-10-12 09:10:00";;126,8
"2020-10-12 09:15:00";;137,6
"2020-10-12 09:20:00";;152,8
"2020-10-12 09:25:00";;177,6
"2020-10-12 09:30:00";;189,8
"2020-10-12 09:35:00";;209
"2020-10-12 09:40:00";;239,8
"2020-10-12 09:45:00";;269,8
"2020-10-12 09:50:00";;281,6
"2020-10-12 09:55:00";;290,4
"2020-10-12 10:00:00";27761050;334
"2020-10-12 10:05:00";;437,2
"2020-10-12 10:10:00";;459,8
"2020-10-12 10:15:00";;445
"2020-10-12 10:20:00";;441,6
"2020-10-12 10:25:00";;412,6
"2020-10-12 10:30:00";;394
"2020-10-12 10:35:00";;383,6
"2020-10-12 10:40:00";;344
"2020-10-12 10:45:00";;366,8
"2020-10-12 10:50:00";;423,8
"2020-10-12 10:55:00";;440,6
"2020-10-12 11:00:00";40317748;407,6
"2020-10-12 11:05:00";;430,2
"2020-10-12 11:10:00";;442,6
"2020-10-12 11:15:00";;441,8
"2020-10-12 11:20:00";;496,8
"2020-10-12 11:25:00";;517,6
"2020-10-12 11:30:00";;598,2
"2020-10-12 11:35:00";;670,4
"2020-10-12 11:40:00";;687,6
"2020-10-12 11:45:00";;785
"2020-10-12 11:50:00";;754,6
"2020-10-12 11:55:00";;740
"2020-10-12 12:00:00";65470464;823,4
"2020-10-12 12:05:00";;949,4
"2020-10-12 12:10:00";;1296
"2020-10-12 12:15:00";;864,4
"2020-10-12 12:20:00";;874,8
"2020-10-12 12:25:00";;849,6
"2020-10-12 12:30:00";;878,4
"2020-10-12 12:35:00";;706,6
"2020-10-12 12:40:00";;730,8
"2020-10-12 12:45:00";;969
"2020-10-12 12:50:00";;993,2
"2020-10-12 12:55:00";;1094,2
"2020-10-12 13:00:00";151807584;2166,2
"2020-10-12 13:05:00";;1964,8
"2020-10-12 13:10:00";;1496,2
"2020-10-12 13:15:00";;1260,4
"2020-10-12 13:20:00";;2132
"2020-10-12 13:25:00";;1696,6
"2020-10-12 13:30:00";;3477
"2020-10-12 13:35:00";;1930,2
"2020-10-12 13:40:00";;2913,2
"2020-10-12 13:45:00";;3150,6
"2020-10-12 13:50:00";;2545,5
"2020-10-12 13:55:00";;1801,4
"2020-10-12 14:00:00";127231592;1813,4
"2020-10-12 14:05:00";;1261,4
"2020-10-12 14:10:00";;1558,4
"2020-10-12 14:15:00";;1708,4
"2020-10-12 14:20:00";;2156,8
"2020-10-12 14:25:00";;3474,6
"2020-10-12 14:30:00";;1755,6
"2020-10-12 14:35:00";;2192
"2020-10-12 14:40:00";;1248,2
"2020-10-12 14:45:00";;1279,6
"2020-10-12 14:50:00";;1879,8
"2020-10-12 14:55:00";;1971,4
"2020-10-12 15:00:00";;3079,2

Frank18000, 2020-10-12:
this from my p1 meter

"DateTime";"Verbruik 1";"Verbruik 2";"Levering 1";"Levering 2"
"2020-10-12 12:05:00";0;972;0;180
"2020-10-12 12:10:00";0;816;0;336
"2020-10-12 12:15:00";0;24;0;72
"2020-10-12 12:20:00";0;48;0;180
"2020-10-12 12:25:00";0;0;0;204
"2020-10-12 12:30:00";0;0;0;288
"2020-10-12 12:35:00";0;1596;0;12
"2020-10-12 12:40:00";0;420;0;132
"2020-10-12 12:45:00";0;576;0;216
"2020-10-12 12:50:00";0;682;0;88
"2020-10-12 12:55:00";0;492;0;396
"2020-10-12 13:00:00";0;77;0;1309
"2020-10-12 13:05:00";0;0;0;1572
"2020-10-12 13:10:00";0;0;0;720
"2020-10-12 13:15:00";0;36;0;648
"2020-10-12 13:20:00";0;0;0;1632
"2020-10-12 13:25:00";0;0;0;1188
"2020-10-12 13:30:00";0;0;0;2436
"2020-10-12 13:35:00";0;0;0;1380
"2020-10-12 13:40:00";0;0;0;1991
"2020-10-12 13:45:00";0;0;0;3348
"2020-10-12 13:50:00";0;0;0;2508
"2020-10-12 13:55:00";0;0;0;1012
"2020-10-12 14:00:00";0;0;0;1296
"2020-10-12 14:05:00";0;0;0;792
"2020-10-12 14:10:00";0;0;0;1032
"2020-10-12 14:15:00";0;0;0;1236
"2020-10-12 14:20:00";0;0;0;1548
"2020-10-12 14:25:00";0;0;0;2796
"2020-10-12 14:30:00";0;0;0;1320
"2020-10-12 14:35:00";0;0;0;1632
"2020-10-12 14:40:00";0;33;0;627
"2020-10-12 14:45:00";0;0;0;660
"2020-10-12 14:50:00";0;0;0;1236
"2020-10-12 14:55:00";0;0;0;1474
"2020-10-12 15:00:00";0;0;0;2304
"2020-10-12 15:05:00";0;0;0;2409
"2020-10-12 15:10:00";0;0;0;2321

addiejanssen, 2020-10-12:
Thanks for that @Frank18000.

The first export you shared looks like the export of a day, not a week.
The week does not have "Energie opgewekt" and "Vermogen gegenereerd", but only one of them.

I think I start to get an idea what is going on, but to be sure, it would help if you can share that Week export as well.

Frank18000, 2020-10-12:
Hi Addie,

i don't have an complete week rapport because the panals became active last saterday

addiejanssen, 2020-10-12:
Ahaa....

But then the numbers make no sense at all and the problem might not be with the plugin, but values that come from the inverter itself. Let's check that first.

In order to use the plugin, you had to install the solaredge_modbus library.
What we need to do, is run the example that is provided by the developer of that library.
This one:
https://github.com/nmakel/solaredge_modbus/blob/master/example.py

To run it, you would do something like:
`
./example.py 192.168.178.241 502
`

It will output something along the line of:
``` Text
Model: SE7K
Type: Three Phase Inverter
Version: 0003.2251
Serial: 7E1AA062
Status: Producing
Temperature: 34.46°C
Current: 2.09A
Phase 1 Current: 0.69A
Phase 2 Current: 0.68A
Phase 3 Current: 0.71A
Phase 1 voltage: 401.30V
Phase 2 voltage: 402.00V
Phase 3 voltage: 404.40V
Phase 1-N voltage: 231.50V
Phase 2-N voltage: 231.30V
Phase 3-N voltage: 232.70V
Frequency: 50.00Hz
Power: 298.00W
Power (Apparent): 483.60VA
Power (Reactive): -381.00VAr
Power Factor: -61.46%
Total Energy: 11373201Wh
DC Current: 0.40A
DC Voltage: 747.00V
DC Power: 302.50W
```

I am really interested to see what your output looks like.

Can you make that work?

Frank18000, 2020-10-12:
i did instal the library 
All the other info looks allright except kWh. I don't know how to install it

mcmellow, 2020-10-13:
Hi Addie,

Installed the plugin yesterday and I receive odd figures too for Total Energy. They does not mact with the SolarEdge-App or website.  A screeshot of the devices is added. I try to run the example.py but it halted with an error on line 43.
[image]

addiejanssen, 2020-10-14:
Hi @Frank18000  and @mcmellow,

Still trying to find out what is going on, but it's not that easy, since I do not have the issue over here...

@mcmellow, thanks for the info. It for sure helps.
Those values are exactly the same as what the example.py would/should return and they are the input for the plugin.

After a day of reading, debugging and fumbling around in the Domoticz code, it is clear to me that lot's of people strugle with the name of the device ("kWh") and what the unit of the counter should be ("Wh" or "kWh").
Looking at the source code in the Domoticz GitHub repo, I see divisions and multiplications of a factor 1.000, but I also see comments along the line of "I assume this should be in kWh". It is just not clear what it should be....

Looking at our setup, the values that are returned by the Docoticz images and exports to csv and xls, all lign up with the information on the SolarEdge monitoring website. So, for me it is working perfectly and based on that, I think that it should upload "Wh" and not "kWh".

Based on what you reported, I wonder if the values that are returned by the inverter (and all the software involved) is correct.
To find that out, we would need a sequence of values.
I executed a  `cat domoticz.log | grep energy_total\": | uniq | tail -n 10` on my domoticz log file and that returned this:

```
    "energy_total": 11393540,
    "energy_total": 11393541,
    "energy_total": 11393542,
    "energy_total": 11393543,
    "energy_total": 11393544,
    "energy_total": 11393545,
    "energy_total": 11393546,
    "energy_total": 11393547,
    "energy_total": 11393548,
    "energy_total": 11393549,
```

Which makes perfect sense: over time, the value goes up, but there are no big jumps in the number.
Our panels have been installed back in 2018 and according to the SolarEdge monitoring site, we have produced 11,39 MWh.
Putting the dots in the last number: 11.393.549 Wh; that's a perfect match.

What would be of great help is:
- getting the last 10 or so unique values from your log, like I did here
- get the total energy produced as reported by the inverter itself; ours has buttons and a display and it reports the value there....
- get the total energy produced as reported on the SolarEdge monitoring website.

Next to that:
- which operating system are you running Domoticz on?
- which version of Domoticz are you using?
- what is the version of the solaredge_modbus library (and the other components it needs) you are using (run: `sudo pip3 list`)

Since the modbus library that is used by the plugin is getting frequent updates in the last weeks, I wonder if that is causing the issues. If that is the case, it would be of great help to downgrade the module and test with the version that was available when I released the plugin. To do so:
```
sudo pip3 uninstall solaredge_modbus
sudo pip3 install solaredge-modbus==0.5.0
```

It's dark now, so the inverter has gone to bed and does not want to play anymore.
I will upgrade the modbus library tomorrow on my development server and see what it the result of that is as well over here.

mcmellow, 2020-10-14:
Hi @Addie

Thanks for your reply. My Rpi crashed last week. Yesterday I have received a new one and installed everything from scratch.  There is history in and saved image of the SD card.

My RPi 3B+ ist at Buster and upgraded yesterdaay. My domoticz is at Version: 2020.2 (build 12467) This is all up to date. My inverter is sleeping as well. Nice moment to stop the plugin and downgrade the solaredge_modbus library to 0.5.0  Let's see what this will do tomorrow.

The total-energy device is an odd device. I have had some discussions about it with GizMoCuz with no result.  There is hardly any documentation on how this device works and calculates. You can set it to device or to computed.  In the setup/setting tab meters/counters you can set dividers as well. It is not clear how this works.  

According to the solaredge-app I am  at 10.52MWh  The log output gives the same amount.
```
pi@mcmellow-pi:/var/log $ cat domoticz.log | grep energy_total\": | uniq | tail -n 20
    "energy_total": 10518359,
    "energy_total": 10518360,
    "energy_total": 10518361,
    "energy_total": 10518362,
    "energy_total": 10518363,
    "energy_total": 10518364,
    "energy_total": 10518365,
    "energy_total": 10518366,
    "energy_total": 10518367,
    "energy_total": 10518368,
    "energy_total": 10518369,
    "energy_total": 10518370,
    "energy_total": 10518371,
    "energy_total": 10518372,
    "energy_total": 10518373,
    "energy_total": 10518374,
    "energy_total": 10518375,
    "energy_total": 10518376,
    "energy_total": 10518377,
    "energy_total": 10518378,

```

Frank18000, 2020-10-15:
my domoticz is the same as MCMellow and also on a raspberry pi
on solar i produced 39,1 kWh in total in domoticz 2581,2 kWh so in my system there is more than a wrong decimal 
The log i cannot reach

addiejanssen, 2020-10-15:
Hi @Frank18000,

There is another way to achieve the same: go to setup, hardware, select the inverter and enable debugging.
Then go to setup, log and add the word `energy` in the filter field on the top right corner.

It will then list the last 2 - 3 updates.

[image]

Keep an eye on the log for a few minutes. What we should see is that the number slowly goes up and does not jump up and down. And the number should be close to that 39.1 kWh reading (if that is correct).

Just to make sure: how did you get to that number of 39.1 kWh?

addiejanssen, 2020-10-15:
Hi @mcmellow,

Those numbers look good.
But what was then reportered in Domoticz at that point?
Or is the problem gone now?

The settings of the device should look like this:
[image]

Frank18000, 2020-10-15:
Hi Addie,

This is wat i get.

When i look at the monitoring site of solar is see 47,74 kWh and when i log i with the solar setapp i see the same value.

greet

Frank

2020-10-15 17:44:10.161 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, <plugin.Average object at 0x6e5d5e50>]
2020-10-15 17:44:10.161 (Zonnepanelen - Total Energy) Updating device from 0:'232.40;3128177459.2' to have values 0:'239.00;3128439603.2'.

2020-10-15 17:45:10.171 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, <plugin.Average object at 0x6e5d5e50>]
2020-10-15 17:45:10.172 (Zonnepanelen - Total Energy) Updating device from 0:'239.00;3128439603.2' to have values 0:'243.20;3128714854.4'.

2020-10-15 17:47:11.491 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, <plugin.Average object at 0x6e5d5e50>]
2020-10-15 17:47:11.492 (Zonnepanelen - Total Energy) Updating device from 0:'242.80;3128990105.6' to have values 0:'236.20;3129265356.8'.

2020-10-15 17:48:11.421 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, <plugin.Average object at 0x6e5d5e50>]
2020-10-15 17:48:11.422 (Zonnepanelen - Total Energy) Updating device from 0:'236.20;3129265356.8' to have values 0:'224.00;3129527500.8'.

2020-10-15 17:49:11.464 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, <plugin.Average object at 0x6e5d5e50>]
2020-10-15 17:49:11.465 (Zonnepanelen - Total Energy) Updating device from 0:'224.00;3129527500.8' to have values 0:'208.80;3129763430.4'.

2020-10-15 17:50:11.452 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, <plugin.Average object at 0x6e5d5e50>]
2020-10-15 17:50:11.452 (Zonnepanelen - Total Energy) Updating device from 0:'208.80;3129763430.4' to have values 0:'191.40;3129986252.8'.

2020-10-15 17:51:11.467 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, <plugin.Average object at 0x6e5d5e50>]
2020-10-15 17:51:11.467 (Zonnepanelen - Total Energy) Updating device from 0:'191.40;3129986252.8' to have values 0:'173.40;3130195968.0'.

mcmellow, 2020-10-15:
Hello Addie.

On the Total Energy devicein Domoticz. During the day the figure today energy is different then the figure on the SolarEdge website or app for today production.  However at the end ot the day, when the inverter is sleeping, the figure for today is on both the same. Domoticz and the SolarEdge website.

addiejanssen, 2020-10-16:
Hi @mcmellow,

That makes sense; the inverter will upload its information in (I think 5 minutes) steps to the SolarEdge servers.
And there can be a delay between that and the availability of the info in the app.

The plugin collects at the intervals that you define yourself.
Domoticz will hang on the last updated value in a 5 minute period.

Since these timeslots probably don't align, you will see slight differences.

At the end of the day, the totals should be more or less the same.
Most days, the delta is less than a few hundred Watt hours; maximum I saw so far was 800 Wh delta between the 2.
When looking at longer intervals (weeks or months) that averages out again and the difference is hardly anything.

addiejanssen, 2020-10-16:
Hi @Frank18000,

Well, the good news is that the numbers go up and don't jump around.
The bad news is that the numbers make no sense...

Look at the last one for example: 3129986252.8
What's strange about this number is the "." in it.
The inverter returns a whole number, so it shouldn't have a . in it.

I had a peek at the code and noticed that I have an error in the inverter tables.
The version you are using is calculating an average of the last so many samples.
That shouldn't be the case and could cause this; it has no real impact, but it shouldn't be like that.

I have updated the plugin to correct this (so we now have version 1.0.5).
Would you be so kind to update your plugin and have a look at the log one more time?
It should not have a "." in those samples anymore.
This will not solve your issue, but should rule out that the "." is a result of calculating an average.

I don't see the relationship between the reported value by the inverter (3129986252.8) and what you reported from the SolarEdge site (47,74 kWh). That doesn't make sense at all....
You reported that this was a new installation; if that is the case, then the 47.74 kWh looks okay, but the 3129986252.8 is really strange.

Is this a (factory) new inverter or is it a used (pre-owned) one?
That number (3129986252.8) is way to big for a brand new inverter.
The only explenation that I can think of, would be that the inverter was not set to 0 when it started up.
That is something that probably can be corrected by doing a factory reset and configure it again.

Domoticz (and probably also SolarEdge) calculates the difference between the values reported by the inverter.
So, let's see if that makes sense then.

Looking at your log:
2020-10-15 17:44:10 3128439603.2
2020-10-15 17:51:11 3130195968.0

The delta in those 7 minutes is then: 1756364.8
Divide that by 7 minutes, that would be (rounded) 250909 per minute.
In that timeframe, your inverter reported about 239W - 173W of power production per minute.
So if we divide that 250909 by 1000, then it could be read as 250.909 or 250W per minute.
That's pretty close again....

What we should do, is get the numbers by the end of today and compare them with the numbers you shared yesterday.
Then we would need to know what the inverter generated today and see if we can get that value by comparing it to the samples from the inverter.

So, can you share:
- the same log entries like you did yesterday of the end of today?
- a screenshot of the Domoticz total energy graphs?
- the generated power as reported by SolarEdge for the timeframe between yesterday and today?

I am also interested in seeing a screenshot of your Setup -> Settings -> Meters/Counters:
[image]

If you can share a screenshot of that too, then I think we have all the information to do the next step.

Thanks,

Addie

Frank18000, 2020-10-16:
Hi Addie,

updated the plug in and his is what i get now.
2020-10-16 14:19:22.252 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, None]
2020-10-16 14:19:22.252 (Zonnepanelen - Total Energy) Updating device from 0:'3358.00;3654549504' to have values 0:'3359.33;3658350592'.

The settings in the screen for meters are exactlly what i have. The inverter is new and i did reset the counters at startup

greet 

Frank

addiejanssen, 2020-10-16:
Great, that's better.
Looks like the "." was caused by the plugin.

Still interested in:
- the same log entries like you did yesterday of the end of today?
- a screenshot of the Domoticz total energy graphs?
- the generated power as reported by SolarEdge for the timeframe between yesterday and today?

Frank18000, 2020-10-16:
ik zal ze straks maken als de panelen geen energie meer produceren. Wat me wel opvalt nu is dat er een factor 1,52 zit tussen de waardes in domoticz en de solar website

Frank18000, 2020-10-16:
2020-10-16 19:38:51.350 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, None]

[image]
[image]
[image]

addiejanssen, 2020-10-16:
Hi @Frank18000,

You missed the log entry that lists the last value reported by the inverter :-)
We're looking for an updated version of this kind of entry:
2020-10-15 17:51:11.467 (Zonnepanelen - Total Energy) Updating device from 0:'191.40;3129986252.8' to have values 0:'173.40;3130195968.0'.
Would you be so kind to look at it early tomorrow morning?

Frank18000, 2020-10-18:
solar edge monitoring 69,54kWh

2020-10-18 08:02:50.961 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, None]
2020-10-18 08:03:50.966 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, None]

2020-10-18 08:27:51.337 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, None]
2020-10-18 08:27:51.338 (Zonnepanelen - Total Energy) Updating device from 0:'0.00;264896513' to have values 0:'0.40;264896513'.

addiejanssen, 2020-10-19:
@Frank18000, I have no clue....

On the 16th, your inverter reported 3 691 970 560 and on the 18th it shows 264 896 513.
I have no idea what is going on over there. These numbers make no sense to me at all.

I have added a new setting in the hardware settings of the plugin:

[image]

There are 2 options here: "Normal" and "Extra".
When you select "Extra" and disable debug, you will get these kind of messages in the log:
```
2020-10-19 14:46:32.457 (inverter) inverter values: {
2020-10-19 14:46:32.457 "c_manufacturer": "SolarEdge",
2020-10-19 14:46:32.457 "c_model": "SE7K",
2020-10-19 14:46:32.457 "c_version": "0003.2251",
2020-10-19 14:46:32.457 "c_deviceaddress": 1,
2020-10-19 14:46:32.457 "c_sunspec_did": 103,
2020-10-19 14:46:32.457 "current": 452,
2020-10-19 14:46:32.457 "p1_current": 150,
2020-10-19 14:46:32.457 "p2_current": 149,
2020-10-19 14:46:32.457 "p3_current": 152,
2020-10-19 14:46:32.457 "current_scale": -2,
2020-10-19 14:46:32.457 "p1_voltage": 4023,
2020-10-19 14:46:32.457 "p2_voltage": 4025,
2020-10-19 14:46:32.457 "p3_voltage": 4051,
2020-10-19 14:46:32.457 "p1n_voltage": 2334,
2020-10-19 14:46:32.457 "p2n_voltage": 2322,
2020-10-19 14:46:32.457 "p3n_voltage": 2321,
2020-10-19 14:46:32.457 "voltage_scale": -1,
2020-10-19 14:46:32.457 "power_ac": 10240,
2020-10-19 14:46:32.457 "power_ac_scale": -1,
2020-10-19 14:46:32.457 "frequency": 5002,
2020-10-19 14:46:32.457 "frequency_scale": -2,
2020-10-19 14:46:32.457 "power_apparent": 10517,
2020-10-19 14:46:32.457 "power_apparent_scale": -1,
2020-10-19 14:46:32.457 "power_reactive": -24000,
2020-10-19 14:46:32.457 "power_reactive_scale": -2,
2020-10-19 14:46:32.457 "power_factor": -9733,
2020-10-19 14:46:32.457 "power_factor_scale": -2,
2020-10-19 14:46:32.457 "energy_total": 11424774,
2020-10-19 14:46:32.457 "energy_total_scale": 0,
2020-10-19 14:46:32.457 "current_dc": 13904,
2020-10-19 14:46:32.457 "current_dc_scale": -4,
2020-10-19 14:46:32.457 "voltage_dc": 7476,
2020-10-19 14:46:32.457 "voltage_dc_scale": -1,
2020-10-19 14:46:32.457 "power_dc": 10395,
2020-10-19 14:46:32.457 "power_dc_scale": -1,
2020-10-19 14:46:32.457 "temperature": 3419,
2020-10-19 14:46:32.457 "temperature_scale": -2,
2020-10-19 14:46:32.457 "status": 4,
2020-10-19 14:46:32.457 "vendor_status": 0
2020-10-19 14:46:32.457 }
```

Would you be so kind to execute a `pip3 list` on your computer.
The list should have this line:
```
solaredge-modbus       0.6.2
```
When it has a lower version, please run `pip3 install solaredge-modbus==0.6.2`.
This will enforce the installation of the latest version.

Would you then be so kind to upgrade the plugin (new version is now 1.0.6), set the log level to extra, set debug to false and then capture these values as returned by your inverter?

Frank18000, 2020-10-20:
i got this

2020-10-20 11:58:37.631 (Zonnepanelen) -> found in Devices
2020-10-20 11:58:37.631 (Zonnepanelen) -> calculating...
2020-10-20 11:58:37.631 (Zonnepanelen) Average: 292.73333333333335 - 15 values
2020-10-20 11:58:37.631 (Zonnepanelen) value = 292.73333333333335
2020-10-20 11:58:37.631 (Zonnepanelen) -> no prepend
2020-10-20 11:58:37.631 (Zonnepanelen) sValue = 292.73
2020-10-20 11:58:37.631 (Zonnepanelen - Power) Updating device from 0:'293.14' to have values 0:'292.73'.
2020-10-20 11:58:37.644 (Zonnepanelen) [<Unit.FREQUENCY: 14>, 'Frequency', 243, 31, 0, {'Custom': '1;Hz'}, 'frequency', 'frequency_scale', '{:.2f}', None, None, <plugin.Average object at 0x6eed6dd0>]
2020-10-20 11:58:37.644 (Zonnepanelen) -> found in Devices
2020-10-20 11:58:37.644 (Zonnepanelen) -> calculating...
2020-10-20 11:58:37.644 (Zonnepanelen) Average: 50.004666666666665 - 15 values
2020-10-20 11:58:37.644 (Zonnepanelen) value = 50.004666666666665
2020-10-20 11:58:37.645 (Zonnepanelen) -> no prepend
2020-10-20 11:58:37.645 (Zonnepanelen) sValue = 50.00
2020-10-20 11:58:37.645 (Zonnepanelen) [<Unit.POWER_APPARENT: 15>, 'Power (Apparent)', 243, 31, 0, {'Custom': '1;VA'}, 'power_apparent', 'power_apparent_scale', '{:.2f}', None, None, <plugin.Average object at 0x6eed6e30>]
2020-10-20 11:58:37.645 (Zonnepanelen) -> found in Devices
2020-10-20 11:58:37.645 (Zonnepanelen) -> calculating...
2020-10-20 11:58:37.645 (Zonnepanelen) Average: 534.5333333333333 - 15 values
2020-10-20 11:58:37.645 (Zonnepanelen) value = 534.5333333333333
2020-10-20 11:58:37.646 (Zonnepanelen) -> no prepend
2020-10-20 11:58:37.646 (Zonnepanelen) sValue = 534.53
2020-10-20 11:58:37.646 (Zonnepanelen - Power (Apparent)) Updating device from 0:'534.66' to have values 0:'534.53'.
2020-10-20 11:58:37.658 (Zonnepanelen) [<Unit.POWER_REACTIVE: 16>, 'Power (Reactive)', 243, 31, 0, {'Custom': '1;VAr'}, 'power_reactive', 'power_reactive_scale', '{:.2f}', None, None, <plugin.Average object at 0x6eed6e50>]
2020-10-20 11:58:37.659 (Zonnepanelen) -> found in Devices
2020-10-20 11:58:37.659 (Zonnepanelen) -> calculating...
2020-10-20 11:58:37.659 (Zonnepanelen) Average: -447.26666666666665 - 15 values
2020-10-20 11:58:37.659 (Zonnepanelen) value = -447.26666666666665
2020-10-20 11:58:37.659 (Zonnepanelen) -> no prepend
2020-10-20 11:58:37.659 (Zonnepanelen) sValue = -447.27
2020-10-20 11:58:37.660 (Zonnepanelen - Power (Reactive)) Updating device from 0:'-447.14' to have values 0:'-447.27'.
2020-10-20 11:58:37.672 (Zonnepanelen) [<Unit.POWER_FACTOR: 17>, 'Power Factor', 243, 6, 0, {}, 'power_factor', 'power_factor_scale', '{:.2f}', None, None, <plugin.Average object at 0x6eed6f10>]
2020-10-20 11:58:37.672 (Zonnepanelen) -> found in Devices
2020-10-20 11:58:37.672 (Zonnepanelen) -> calculating...
2020-10-20 11:58:37.672 (Zonnepanelen) Average: -54.43933333333334 - 15 values
2020-10-20 11:58:37.672 (Zonnepanelen) value = -54.43933333333334
2020-10-20 11:58:37.672 (Zonnepanelen) -> no prepend
2020-10-20 11:58:37.672 (Zonnepanelen) sValue = -54.44
2020-10-20 11:58:37.673 (Zonnepanelen - Power Factor) Updating device from 0:'-54.50' to have values 0:'-54.44'.
2020-10-20 11:58:37.685 (Zonnepanelen) [<Unit.ENERGY_TOTAL: 18>, 'Total Energy', 243, 29, 4, {}, 'energy_total', 'energy_total_scale', '{};{}', <Unit.POWER_AC: 13>, None, None]
2020-10-20 11:58:37.685 (Zonnepanelen) -> found in Devices
2020-10-20 11:58:37.685 (Zonnepanelen) -> calculating...
2020-10-20 11:58:37.685 (Zonnepanelen) value = 1054408705
2020-10-20 11:58:37.685 (Zonnepanelen) -> has prepend
2020-10-20 11:58:37.686 (Zonnepanelen) prepend = 292.73
2020-10-20 11:58:37.686 (Zonnepanelen) sValue = 292.73;1054408705
2020-10-20 11:58:37.686 (Zonnepanelen - Total Energy) Updating device from 0:'293.14;1054408705' to have values 0:'292.73;1054408705'.
2020-10-20 11:58:37.703 (Zonnepanelen) [<Unit.CURRENT_DC: 19>, 'DC Current', 243, 23, 0, {}, 'current_dc', 'current_dc_scale', '{:.2f}', None, None, <plugin.Average object at 0x6eed6ed0>]
2020-10-20 11:58:37.703 (Zonnepanelen) -> found in Devices
2020-10-20 11:58:37.703 (Zonnepanelen) -> calculating...
2020-10-20 11:58:37.703 (Zonnepanelen) Average: 0.3975066666666667 - 15 values
2020-10-20 11:58:37.703 (Zonnepanelen) value = 0.3975066666666667
2020-10-20 11:58:37.703 (Zonnepanelen) -> no prepend
2020-10-20 11:58:37.703 (Zonnepanelen) sValue = 0.40
2020-10-20 11:58:37.704 (Zonnepanelen) [<Unit.VOLTAGE_DC: 20>, 'DC Voltage', 243, 8, 0, {}, 'voltage_dc', 'voltage_dc_scale', '{:.2f}', None, None, <plugin.Average object at 0x6eed6ef0>]
2020-10-20 11:58:37.704 (Zonnepanelen) -> found in Devices
2020-10-20 11:58:37.704 (Zonnepanelen) -> calculating...
2020-10-20 11:58:37.704 (Zonnepanelen) Average: 747.4866666666668 - 15 values
2020-10-20 11:58:37.704 (Zonnepanelen) value = 747.4866666666668
2020-10-20 11:58:37.704 (Zonnepanelen) -> no prepend
2020-10-20 11:58:37.704 (Zonnepanelen) sValue = 747.49
2020-10-20 11:58:37.704 (Zonnepanelen - DC Voltage) Updating device from 0:'747.44' to have values 0:'747.49'.
2020-10-20 11:58:37.720 (Zonnepanelen) [<Unit.POWER_DC: 21>, 'DC Power', 248, 1, 0, {}, 'power_dc', 'power_dc_scale', '{:.2f}', None, None, <plugin.Average object at 0x6eed6f30>]
2020-10-20 11:58:37.720 (Zonnepanelen) -> found in Devices
2020-10-20 11:58:37.720 (Zonnepanelen) -> calculating...
2020-10-20 11:58:37.720 (Zonnepanelen) Average: 297.1686666666667 - 15 values
2020-10-20 11:58:37.720 (Zonnepanelen) value = 297.1686666666667
[... cut ...]
2020-10-20 11:58:37.735 (Zonnepanelen) [<Unit.TEMPERATURE: 22>, 'Temperature', 243, 5, 0, {}, 'temperature', 'temperature_scale', '{:.2f}', None, None, <plugin.Maximum object at 0x6eed6f70>]
2020-10-20 11:58:37.735 (Zonnepanelen) -> found in Devices
2020-10-20 11:58:37.735 (Zonnepanelen) -> calculating...
2020-10-20 11:58:37.736 (Zonnepanelen) Maximum: 27.16 - 15 values
2020-10-20 11:58:37.736 (Zonnepanelen) value = 27.16
2020-10-20 11:58:37.736 (Zonnepanelen) -> no prepend
2020-10-20 11:58:37.736 (Zonnepanelen) sValue = 27.16
2020-10-20 11:58:37.736 (Zonnepanelen) Updated 13 values out of 22

addiejanssen, 2020-10-20:
Thanks @Frank18000.

I have logged an issue with the developer of the library that is used by the plugin to get the values from the inverter:
https://github.com/nmakel/solaredge_modbus/issues/10

He may come back with additional questions or things to try.
Would you be so kind to monitor that topic as well?

I think that is our best option to get to the bottom of this strange issue.

Frank18000, 2020-10-20:
thx addie i will monitor this

addiejanssen, 2020-10-21:
@Frank18000,

The developer states that this is related to an issue in an older version of his library.

I would like to verify that statement.
Would you be so kind to execute: `pip3 list` on your computer.
Can you share the resulting list that is produced?

Then the next step is to make sure that you use a version of the library that should not have the issue.
To do that, I would like to install a particular version so we know which one we are looking at.
Could you please run `pip3 install solaredge-modbus==0.6.2`, BUT ONLY AFTER YOU CREATED THAT LIST IN THE PREVIOUS STEP.

Once you have done that, please share the inverter values again; with a bit of luck, the energy_total should be close to what SolarEdge is reporting....

Frank18000, 2020-10-21:
This is what i got

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Oct 20 12:17:22 2020
pi@raspberrypi:~ $ pip3 list
Package           Version
----------------- -----------
appdirs           1.4.3
asn1crypto        0.24.0
astroid           2.1.0
asttokens         1.1.13
automationhat     0.2.0
beautifulsoup4    4.7.1
blinker           1.4
blinkt            0.1.2
buttonshim        0.0.2
Cap1xxx           0.1.3
certifi           2018.8.24
chardet           3.0.4
Click             7.0
colorama          0.3.7
colorzero         1.1
cookies           2.2.1
cryptography      2.6.1
cycler            0.10.0
decorator         4.3.0
docutils          0.14
drumhat           0.1.0
entrypoints       0.3
envirophat        1.0.0
ExplorerHAT       0.4.2
Flask             1.0.2
fourletterphat    0.1.0
gpiozero          1.5.1
guizero           0.6.0
html5lib          1.0.1
idna              2.6
ipykernel         4.9.0
ipython           5.8.0
ipython-genutils  0.2.0
isort             4.3.4
itsdangerous      0.24
jedi              0.13.2
Jinja2            2.10
jupyter-client    5.2.3
jupyter-core      4.4.0
keyring           17.1.1
keyrings.alt      3.1.1
kiwisolver        1.0.1
lazy-object-proxy 1.3.1
logilab-common    1.4.2
lxml              4.3.2
MarkupSafe        1.1.0
matplotlib        3.0.2
mccabe            0.6.1
microdotphat      0.2.1
mote              0.0.4
motephat          0.0.2
mypy              0.670
mypy-extensions   0.4.1
nudatus           0.0.4
numpy             1.16.2
oauthlib          2.1.0
olefile           0.46
pantilthat        0.0.7
parso             0.3.1
pexpect           4.6.0
pgzero            1.2
phatbeat          0.1.1
pianohat          0.1.0
picamera          1.13
pickleshare       0.7.5
picraft           1.0
piglow            1.2.5
pigpio            1.44
Pillow            5.4.1
pip               18.1
prompt-toolkit    1.0.15
psutil            5.5.1
pycairo           1.16.2
pycodestyle       2.4.0
pycrypto          2.6.1
pyflakes          2.0.0
pygame            1.9.4.post1
Pygments          2.3.1
PyGObject         3.30.4
pyinotify         0.9.6
PyJWT             1.7.0
pylint            2.2.2
pymodbus          2.4.0
pyOpenSSL         19.0.0
pyparsing         2.2.0
pyserial          3.4
python-apt        1.8.4.1
python-dateutil   2.7.3
pyxdg             0.25
pyzmq             17.1.2
qtconsole         4.3.1
rainbowhat        0.1.0
requests          2.21.0
requests-oauthlib 1.0.0
responses         0.9.0
roman             2.0.0
RPi.GPIO          0.7.0
RTIMULib          7.2.1
scrollphat        0.0.7
scrollphathd      1.2.1
SecretStorage     2.3.1
semver            2.0.1
Send2Trash        1.5.0
sense-emu         1.1
sense-hat         2.2.0
setuptools        40.8.0
simplegeneric     0.8.1
simplejson        3.16.0
six               1.15.0
skywriter         0.0.7
sn3218            1.2.7
solaredge-modbus  0.6.2
soupsieve         1.8
spidev            3.4
ssh-import-id     5.7
thonny            3.2.6
tornado           5.1.1
touchphat         0.0.1
traitlets         4.3.2
twython           3.7.0
typed-ast         1.3.1
uflash            1.2.4
unicornhathd      0.0.4
urllib3           1.24.1
wcwidth           0.1.7
webencodings      0.5.1
Werkzeug          0.14.1
wheel             0.32.3
wrapt             1.10.11

addiejanssen, 2020-10-21:
Before or after upgrading the solaredge-modbus library?

Frank18000, 2020-10-21:
before updating

addiejanssen, 2020-10-21:
OK, let me report that back to the developer. Thanks.

addiejanssen, 2020-10-21:
While we are waiting for Niels to respond: he stated that some of the problems started with version 0.6.x.
I would like to try to use an older version; the one that I used when I created the plugin.
We can always upgrade again to the latest version; so no worry, this will not break anything.

Can you downgrade the library and see what the outcome of that is?
To do so: `pip3 install solaredge-modbus==0.5.0`.
Then let's see if that version provides different results.

Frank18000, 2020-10-21:
i instald the older version. And restarted Domoticz. Do i have to resart the inverter to?

addiejanssen, 2020-10-21:
There is no need to restart the inverter.
Can you share the inverter values from the log again?

Frank18000, 2020-10-21:
2020-10-21 11:31:09.790 (Zonnepanelen) inverter values: {
2020-10-21 11:31:09.790 "c_model": "SE5K-RW0TEBNN4",
2020-10-21 11:31:09.790 "c_version": "0004.0010.0025",
2020-10-21 11:31:09.790 "c_deviceaddress": 1,
2020-10-21 11:31:09.790 "c_sunspec_did": 103,
2020-10-21 11:31:09.790 "current": 229,
2020-10-21 11:31:09.790 "p1_current": 68,
2020-10-21 11:31:09.790 "p2_current": 77,
2020-10-21 11:31:09.790 "p3_current": 83,
2020-10-21 11:31:09.790 "current_scale": -2,
2020-10-21 11:31:09.790 "p1_voltage": 3976,
2020-10-21 11:31:09.790 "p2_voltage": 3965,
2020-10-21 11:31:09.790 "p3_voltage": 3990,
2020-10-21 11:31:09.790 "p1n_voltage": 2292,
2020-10-21 11:31:09.790 "p2n_voltage": 2304,
2020-10-21 11:31:09.790 "p3n_voltage": 2291,
2020-10-21 11:31:09.790 "voltage_scale": -1,
2020-10-21 11:31:09.790 "power_ac": 3400,
2020-10-21 11:31:09.790 "power_ac_scale": -2,
2020-10-21 11:31:09.790 "frequency": 4997,
2020-10-21 11:31:09.790 "frequency_scale": -2,
2020-10-21 11:31:09.790 "power_apparent": 5241,
2020-10-21 11:31:09.790 "power_apparent_scale": -1,
2020-10-21 11:31:09.790 "power_reactive": -5230,
2020-10-21 11:31:09.790 "power_reactive_scale": -1,
2020-10-21 11:31:09.790 "power_factor": -5259,
2020-10-21 11:31:09.790 "power_factor_scale": -3,
2020-10-21 11:31:09.790 "energy_total": 1178992641,
2020-10-21 11:31:09.790 "energy_total_scale": 0,
2020-10-21 11:31:09.790 "current_dc": 4611,
2020-10-21 11:31:09.790 "current_dc_scale": -5,
2020-10-21 11:31:09.790 "voltage_dc": 7483,
2020-10-21 11:31:09.790 "voltage_dc_scale": -1,
2020-10-21 11:31:09.790 "power_dc": 3451,
2020-10-21 11:31:09.790 "power_dc_scale": -2,
2020-10-21 11:31:09.790 "temperature": 2924,
2020-10-21 11:31:09.790 "temperature_scale": -2,
2020-10-21 11:31:09.790 "status": 4,
2020-10-21 11:31:09.790 "vendor_status": 0
2020-10-21 11:31:09.790 }

addiejanssen, 2020-10-21:
Hi @Frank18000,

Niels responded and he has an interesting point which means that your library is not the latest one.
I have an idea why.
Can you run the following commands on your computer:
```
pip3 uninstall solaredge-modbus
sudo pip3 list
```
And then share the list here.

Then the next step:
```
sudo pip3 install solaredge-modbus==0.6.2
```

Then restart Domoticz (or even better: restart the computer if possible) and share the inverter values again.

addiejanssen, 2020-10-21:
Hi @mcmellow,

We are still working on Franks issue, but it is unclear what the status is of your setup.
Is everything working correct now for you?
If not: what isn't?

mcmellow, 2020-10-21:
> Hi @mcmellow,
> 
> We are still working on Franks issue, but it is unclear what the status is of your setup.
> Is everything working correct now for you?
> If not: what isn't?

It seems to look OK, no errors.  But there are hardly any messages.

Frank18000, 2020-10-22:
I did what you asked. And after uninstal version 0.6.1 came up in the pip3 list. After unstalling 0.6.1 and installing 0.6.2 it looks to be all right :-D

2020-10-22 08:58:45.847 (Zonnepanelen) inverter values: {
2020-10-22 08:58:45.847 "c_manufacturer": "SolarEdge",
2020-10-22 08:58:45.847 "c_model": "SE5K-RW0TEBNN4",
2020-10-22 08:58:45.847 "c_version": "0004.0010.0025",
2020-10-22 08:58:45.847 "c_deviceaddress": 1,
2020-10-22 08:58:45.847 "c_sunspec_did": 103,
2020-10-22 08:58:45.847 "current": 228,
2020-10-22 08:58:45.847 "p1_current": 69,
2020-10-22 08:58:45.847 "p2_current": 77,
2020-10-22 08:58:45.847 "p3_current": 81,
2020-10-22 08:58:45.847 "current_scale": -2,
2020-10-22 08:58:45.847 "p1_voltage": 3991,
2020-10-22 08:58:45.847 "p2_voltage": 3988,
2020-10-22 08:58:45.847 "p3_voltage": 4019,
2020-10-22 08:58:45.847 "p1n_voltage": 2317,
2020-10-22 08:58:45.847 "p2n_voltage": 2305,
2020-10-22 08:58:45.847 "p3n_voltage": 2302,
2020-10-22 08:58:45.847 "voltage_scale": -1,
2020-10-22 08:58:45.847 "power_ac": 4100,
2020-10-22 08:58:45.847 "power_ac_scale": -2,
2020-10-22 08:58:45.847 "frequency": 5000,
2020-10-22 08:58:45.847 "frequency_scale": -2,
2020-10-22 08:58:45.847 "power_apparent": 5246,
2020-10-22 08:58:45.847 "power_apparent_scale": -1,
2020-10-22 08:58:45.847 "power_reactive": -5230,
2020-10-22 08:58:45.847 "power_reactive_scale": -1,
2020-10-22 08:58:45.847 "power_factor": -7625,
2020-10-22 08:58:45.847 "power_factor_scale": -3,
2020-10-22 08:58:45.847 "energy_total": 85332,
2020-10-22 08:58:45.847 "energy_total_scale": 0,
2020-10-22 08:58:45.847 "current_dc": 5567,
2020-10-22 08:58:45.847 "current_dc_scale": -5,
2020-10-22 08:58:45.847 "voltage_dc": 7476,
2020-10-22 08:58:45.847 "voltage_dc_scale": -1,
2020-10-22 08:58:45.847 "power_dc": 4162,
2020-10-22 08:58:45.847 "power_dc_scale": -2,
2020-10-22 08:58:45.847 "temperature": 2591,
2020-10-22 08:58:45.847 "temperature_scale": -2,
2020-10-22 08:58:45.847 "status": 4,
2020-10-22 08:58:45.847 "vendor_status": 0
2020-10-22 08:58:45.847 }

addiejanssen, 2020-10-22:
That's good news @Frank18000.

As Niels stated: there was an issue with his library that got fixed in 0.6.2; the fact that "c_manufacturer" was not in your list, showed that you were not using that version.

Good to see that it is now working.
I can imagine that you now have a messed up display of values for energy_total.
What you could do, is delete the energy_total device and then stop and start the "Zonnepanelen" hardware.
That will re-create the device and it will then start fresh.
Then it would be interesting to see if the graphs are using the correct scaling.

Frank18000, 2020-10-22:
what do yoo mean with start and stop the hardware?

Frank18000, 2020-10-22:
that did the trick of ressetting the mess i had. great and thx

addiejanssen, 2020-10-22:
In here (ignore the arrow; I reused an image):

[image]

Select the "Zonnepalen" in your setup.
Click on "Enable" to disable (what's in a word); update it.
Then select it again and click on "Enable" to enable it again; update it.

That stops and starts the hardware.
The "Add missing devices" setting should be set to "Yes"; that will recreate the device that you deleted.

Frank18000, 2020-10-22:
it works like a charme 👍 :-D

[image]

addiejanssen, 2020-10-22:
It was a bit of a journey to get here, but good to see that it is working now.

closed 2020-10-22 by addiejanssen