addiejanssen/domoticz-solaredge-modbustcp-plugin #5: read meter data

pfolcato, 2020-11-22:
hi, how can I import the data of Meter in Domoticz?
When I execute the Solaredge_Modbus example.py code I get this result.... but I don't know python.
The meter is connected in Export+Import mode so when I'm able to read "power" I know everything (home consumption... export... import)  
in this example I'm importing 1511 Watt 

"meters": {
        "Meter1": {
            "c_manufacturer": "SolarEdge",
            "c_model": "SE-MTR-3Y-400V-A",
            "c_option": "Export+Import",
            "c_version": "74",
            "c_serialnumber": "xxxxxxxx",
            "c_deviceaddress": 2,
            "c_sunspec_did": 203,
            "current": 68,
            "p1_current": 68,
            "p2_current": 0,
            "p3_current": 0,
            "current_scale": -1,
            "voltage_ln": 24297,
            "p1n_voltage": 24297,
            "p2n_voltage": 41,
            "p3n_voltage": 0,
            "voltage_ll": 0,
            "p12_voltage": 0,
            "p23_voltage": 0,
            "p31_voltage": 0,
            "voltage_scale": -2,
            "frequency": 5000,
            "frequency_scale": -2,
            "power": -1511,
            "p1_power": -1511,
            "p2_power": 0,
            "p3_power": 0,
            "power_scale": 0,
            "power_apparent": 1540,
            "p1_power_apparent": 1539,
            "p2_power_apparent": 0,
            "p3_power_apparent": 0,
            "power_apparent_scale": 0,
            "power_reactive": -295,
            "p1_power_reactive": -295,
            "p2_power_reactive": 0,
            "p3_power_reactive": 0,
            "power_reactive_scale": 0,
            "power_factor": 9736,
            "p1_power_factor": 9210,
            "p2_power_factor": 10000,
            "p3_power_factor": 10000,
            "power_factor_scale": -2,
            "export_energy_active": 21,
            "p1_export_energy_active": 21,
            "p2_export_energy_active": 0,
            "p3_export_energy_active": 0,
            "import_energy_active": 12237,
            "p1_import_energy_active": 12237,
            "p2_import_energy_active": 0,
            "p3_import_energy_active": 0,
            "energy_active_scale": 0,
            "export_energy_apparent": 0,
            "p1_export_energy_apparent": 0,
            "p2_export_energy_apparent": 0,
            "p3_export_energy_apparent": 0,
            "import_energy_apparent": 0,
            "p1_import_energy_apparent": 0,
            "p2_import_energy_apparent": 0,
            "p3_import_energy_apparent": 0,
            "energy_apparent_scale": -32768,
            "import_energy_reactive_q1": 0,
            "p1_import_energy_reactive_q1": 0,
            "p2_import_energy_reactive_q1": 0,
            "p3_import_energy_reactive_q1": 0,
            "import_energy_reactive_q2": 0,
            "p1_import_energy_reactive_q2": 0,
            "p2_import_energy_reactive_q2": 0,
            "p3_import_energy_reactive_q2": 0,
            "export_energy_reactive_q3": 0,
            "p1_export_energy_reactive_q3": 0,
            "p2_export_energy_reactive_q3": 0,
            "p3_export_energy_reactive_q3": 0,
            "export_energy_reactive_q4": 0,
            "p1_export_energy_reactive_q4": 0,
            "p2_export_energy_reactive_q4": 0,
            "p3_export_energy_reactive_q4": 0,
            "energy_reactive_scale": -32768
        }
    },
    "batteries": {}
}

addiejanssen, 2020-11-23:
Hi pfolcato,

Can you provide a bit more background?
Did you install the plugin already?
Was the plugin able to communicate with your inverter (it will report that in the log)?
Did the plugin install the devices in Domoticz?
When the plugin is failing, can you enable debugging and share the Domoticz log entries?

pfolcato, 2020-11-23:
hi, the plugin work very well! everything is working, but I need to read another value from inverter.
I have smartmeter connected to Inverter by rs485 and I want to read the "power" table reported in previus post as "power": -1511,
Your plugin read only Inverter data but no smartmeter or battery.
For read smarmeter value I need to add at your plugin this example from sunspec_monitor

>>> inverter.meters()
    {
        'Meter1': Meter1(10.0.0.123:1502, connectionType.TCP: timeout=1, retries=3, unit=0x1)
    }

    >>> meter1 = inverter.meters()["Meter1"]
    >>> meter1
    Meter1(10.0.0.123:1502, connectionType.TCP: timeout=1, retries=3, unit=0x1)

    >>> meter1.read_all()
    {
        'c_manufacturer': 'SolarEdge',
        'c_model': 'PRO380-Mod',
        'c_option': 'Export+Import',
        'c_version': '2.19',
        'c_serialnumber': '12312332',
        'c_deviceaddress': 1,
        'c_sunspec_did': 203,
        'current': -13,
        ...
    }
and use the value "power" as another device. 
But I'm not a python programmer.... :(

addiejanssen, 2020-11-23:
OK, now I get your point.

The solaredge_modbus library that is used by the plugin has been updated to support inverters, meters and batteries some time ago. However, the plugin is using an approach before those devices were supported.
To do what you are looking for, would mean a pretty big change to the plugin.
Since I do not have access to the hardware myself, testing it would become a challenge.

I will put this on the "requested features" list, but I can not guarantee that this functionality will be added...

pfolcato, 2020-11-28:
I have to thank you Addiejanssen!
I'm not a programmer but thanks to your plugin i added some code to read the solaredge meter, change the values ​​that are called with the same keys in the inverter and then merge my dictionary variable with your dictionary variable that your plugin processes to assign the read values to the idx of domoticz.
It's not very elegant but it works ;)
If anyone is interested in this gorgeous plugin that allows to create devices in domoticz automatically, read both the inverter and the SolarEdge meter via ModbusTCP, I am attaching the modified plugin.
Thank you again Addiejanssen!
[plugin.zip](https://github.com/addiejanssen/domoticz-solaredge-modbustcp-plugin/files/5611246/plugin.zip)

gropi75, 2020-12-11:
Great work t both of You.
The plugin works well, I copied the modified part to the 3phase section and it is running well.
[image]

pfolcato, 2020-12-11:
hello gropi75 I'm glad you used my small part too! the greatest work was done by the owner of this wonderful script!

gropi75, 2020-12-12:
One update: in case of a 3phase inverter the power will be retured from the meter for each phase and in total. So there is 1 change in the section 3ph needed: use for power just `power` instead of `p1_power`

[plugin.zip](https://github.com/addiejanssen/domoticz-solaredge-modbustcp-plugin/files/5682666/plugin.zip)

closed 2021-04-05 by addiejanssen