addiejanssen/domoticz-solaredge-modbustcp-plugin #6: Plugin on Synology

To0wnn, 2020-11-23:
Hi I want to use your plugin on synology but somehow it is not working.

The following I have in domoticz log:
`2020-11-23 14:09:04.554 Error: (SolarEdge_ModbusTCP) failed to load 'plugin.py', Python Path used was '/usr/local/domoticz/var/plugins/domoticz-solaredge-modbustcp-plugin/:/var/packages/py3k/target/usr/local/lib/python38.zip:/var/packages/py3k/target/usr/local/lib/python3.8:/var/packages/py3k/target/usr/local/lib/python3.8/lib-dynload:/var/packages/py3k/target/usr/local/lib/python3.8/site-packages'.
2020-11-23 14:09:04.555 Error: (SolarEdge) Module Import failed, exception: 'ModuleNotFoundError'
2020-11-23 14:09:04.555 Error: (SolarEdge) Module Import failed: ' Name: solaredge_modbus'
2020-11-23 14:09:04.555 Error: (SolarEdge) Error Line details not available.
2020-11-23 14:10:12.300 Error: SolarEdge hardware (7) thread seems to have ended unexpectedly
2020-11-23 14:10:26.351 Error: SolarEdge hardware (7) thread seems to have ended unexpectedly
2020-11-23 14:10:40.387 Error: SolarEdge hardware (7) thread seems to have ended unexpectedly
2020-11-23 14:10:54.422 Error: SolarEdge hardware (7) thread seems to have ended unexpectedly
2020-11-23 14:11:08.457 Error: SolarEdge hardware (7) thread seems to have ended unexpectedly
2020-11-23 14:11:22.495 Error: SolarEdge hardware (7) thread seems to have ended unexpectedly
2020-11-23 14:11:36.534 Error: SolarEdge hardware (7) thread seems to have ended unexpectedly
2020-11-23 14:11:50.583 Error: SolarEdge hardware (7) thread seems to have ended unexpectedly
2020-11-23 14:12:04.619 Error: SolarEdge hardware (7) thread seems to have ended unexpectedly
2020-11-23 14:12:11.980 Error: (SolarEdge_ModbusTCP) failed to load 'plugin.py', Python Path used was '/usr/local/domoticz/var/plugins/domoticz-solaredge-modbustcp-plugin/:/var/packages/py3k/target/usr/local/lib/python38.zip:/var/packages/py3k/target/usr/local/lib/python3.8:/var/packages/py3k/target/usr/local/lib/python3.8/lib-dynload:/var/packages/py3k/target/usr/local/lib/python3.8/site-packages'.
2020-11-23 14:12:11.980 Error: (SolarEdge) Module Import failed, exception: 'ModuleNotFoundError'
2020-11-23 14:12:11.980 Error: (SolarEdge) Module Import failed: ' Name: solaredge_modbus'
2020-11-23 14:12:11.980 Error: (SolarEdge) Error Line details not available.
2020-11-23 14:13:14.800 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:13:28.846 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:13:42.878 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:14:10.946 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:14:24.982 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:14:39.017 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:14:53.067 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:15:07.162 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:15:21.194 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:15:35.229 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:15:49.266 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:16:03.307 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:16:17.345 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:16:31.402 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:16:45.440 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:16:59.478 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly
2020-11-23 14:17:13.515 Error: SolarEdge hardware (6) thread seems to have ended unexpectedly`

addiejanssen, 2020-11-23:
Hi To0wnn,

The plugin uses the solaredge_modbus library (https://github.com/nmakel/solaredge_modbus) to communicate with the inverter. Looking at your log, that library is not installed.

On Ubuntu, I would do the following to install that library:
```
cd domoticz/plugins/domoticz-solaredge-modbustcp-plugin
sudo pip3 install -r requirements.txt
```
But I have no idea how/if that would work on a Synology NAS.

To0wnn, 2020-11-23:
I already did but somehow it is not working 

```
Requirement already satisfied: solaredge_modbus==0.6.2 in /volume1/@appstore/py3k/usr/local/lib/python3.8/site-packages/solaredge_modbus-0.6.2-py3.8.egg (from -r requirements.txt (line 1)) (0.6.2)
Requirement already satisfied: pymodbus>=2.3.0 in /volume1/@appstore/py3k/usr/local/lib/python3.8/site-packages/pymodbus-2.4.0-py3.8.egg (from solaredge_modbus==0.6.2->-r requirements.txt (line 1)) (2.4.0)
Requirement already satisfied: pyserial>=3.4 in /volume1/@appstore/py3k/usr/local/lib/python3.8/site-packages/pyserial-3.5-py3.8.egg (from pymodbus>=2.3.0->solaredge_modbus==0.6.2->-r requirements.txt (line 1)) (3.5)
Requirement already satisfied: six>=1.15.0 in /volume1/@appstore/py3k/usr/local/lib/python3.8/site-packages/six-1.15.0-py3.8.egg (from pymodbus>=2.3.0->solaredge_modbus==0.6.2->-r requirements.txt (line 1)) (1.15.0)
```

I download solaredge_modbus separately and run example.py it works

addiejanssen, 2020-11-23:
Looking at your initial log entries, the Domotociz runtime is using the following paths to search for the modules:
```
/usr/local/domoticz/var/plugins/domoticz-solaredge-modbustcp-plugin/
/var/packages/py3k/target/usr/local/lib/python38.zip
/var/packages/py3k/target/usr/local/lib/python3.8
/var/packages/py3k/target/usr/local/lib/python3.8/lib-dynload
/var/packages/py3k/target/usr/local/lib/python3.8/site-packages
```

Looking at your last message, the libraries are installed in:
```
/volume1/@appstore/py3k/usr/local/lib/python3.8/site-packages
```

You could try to move these 4 .egg files into one of the folders that Domoticz is using....

To0wnn, 2020-11-24:
Actually the same path all .egg files are standing in the same folder

addiejanssen, 2020-11-24:
As I don't have a Synology NAS available to test this, I did a search in the Domoticz forum:
https://domoticz.com/forum/search.php?keywords=synology+ModuleNotFoundError

Got 5 hits, but this topic looks interesting:
https://domoticz.com/forum/viewtopic.php?f=6&t=34232

I think this is not an issue with this plugin, but its an issue running any plugin on a Synology NAS.
Please have a look at the Domoticz forum.

closed 2020-12-10 by addiejanssen