addiejanssen/domoticz-solaredge-modbustcp-plugin #8: Module import failed

pczolee, 2021-01-25:
Hello.

My system is Debian 9.13 x64. Python 3.5.3.
I followed all your instructions, but the plugin not working :(. In the debug log I found the following:

2021-01-25 17:24:30.628 Error: (SolarEdge_ModbusTCP) failed to load 'plugin.py', Python Path used was '/opt/domoticz/plugins/domoticz-solaredge-modbustcp-plugin/:/usr/lib/python35.zip:/usr/lib/python3.5:/usr/lib/python3.5/plat-x86_64-linux-gnu:/usr/lib/python3.5/lib-dynload'.
2021-01-25 17:24:30.628 Error: (SolarEdge Inverter1) Module Import failed, exception: 'ImportError'
2021-01-25 17:24:30.628 Error: (SolarEdge Inverter1) Module Import failed: ' Name: solaredge_modbus'
2021-01-25 17:24:30.628 Error: (SolarEdge Inverter1) Error Line details not available. 

# pip3 list
acme (0.28.0)
ardexaplugin (0.2.3)
asn1crypto (0.24.0)
bitstring (3.1.7)
certbot (0.28.0)
certifi (2016.2.28)
chardet (3.0.4)
click (7.1.2)
ConfigArgParse (0.11.0)
configobj (5.0.6)
cryptography (2.3)
ecdsa (0.16.1)
esptool (3.0)
future (0.15.2)
idna (2.6)
iotop (0.6)
josepy (1.1.0)
lxml (3.7.1)
mock (2.0.0)
natsort (4.0.3)
netifaces (0.10.4)
parsedatetime (2.4)
pbr (4.2.0)
pip (9.0.1)
pymodbus (2.4.0)
pyModbusTCP (0.1.8)
pyOpenSSL (19.0.0)
pyRFC3339 (1.0)
pyserial (3.5)
python-apt (1.4.3)
pythondialog (3.4.0)
pytz (2016.7)
pyudev (0.21.0)
reedsolo (1.5.4)
requests (2.21.0)
requests-toolbelt (0.7.0)
setuptools (33.1.1)
six (1.15.0)
solaredge-modbus (0.6.4)
urllib3 (1.24.1)
zope.component (4.3.0)
zope.event (4.2.0)
zope.hookable (4.0.4)
zope.interface (4.3.2)

The missing module is installed, so I have no idea now what is the problem.

Thanks

pczolee, 2021-01-26:
Well I partially found the problem. I had old version of python. The solaredge_modbus need python 3.7 . So I built 3.7.3. Now the example.py (from solaredge_modbus) working well:

Registers:
	Manufacturer: SolarEdge
	Model: SE17K-XXXXXXX
	Type: Three Phase Inverter
	Version: 0004.0011.0030
	Serial: XXXXXXX
	Status: Producing
	Temperature: 18.94°C
	Current: 5.67A
	Phase 1 Current: 1.89A
	Phase 2 Current: 1.89A
	Phase 3 Current: 1.89A
	Phase 1 voltage: 404.80V
	Phase 2 voltage: 409.20V
	Phase 3 voltage: 408.40V
	Phase 1-N voltage: 231.00V
	Phase 2-N voltage: 234.40V
	Phase 3-N voltage: 240.50V
	Frequency: 49.97Hz
	Power: 1292.00W
	Power (Apparent): 1338.50VA
	Power (Reactive): -350.00VAr
	Power Factor: -96.48%
	Total Energy: 11273136Wh
	DC Current: 1.75A
	DC Voltage: 748.00V
	DC Power: 1311.60W

But from the domoticz I still have this:

 2021-01-26 10:11:54.031 Error: (SolarEdge_ModbusTCP) failed to load 'plugin.py', Python Path used was '/opt/domoticz/plugins/domoticz-solaredge-modbustcp-plugin/:/usr/lib/python35.zip:/usr/lib/python3.5:/usr/lib/python3.5/plat-x86_64-linux-gnu:/usr/lib/python3.5/lib-dynload'.
2021-01-26 10:11:54.031 Error: (SolarEdge Inverter) Module Import failed, exception: 'ImportError'
2021-01-26 10:11:54.031 Error: (SolarEdge Inverter) Module Import failed: ' Name: solaredge_modbus'
2021-01-26 10:11:54.031 Error: (SolarEdge Inverter) Error Line details not available. 

I changed the first line in plugin.py 
from:
#!/usr/bin/env python
to:
#!/usr/local/bin/python3

But nothing changed.

If i run it from the terminal this is the output:

python3 plugin.py 
Traceback (most recent call last):
  File "plugin.py", line 48, in <module>
    import Domoticz
ModuleNotFoundError: No module named 'Domoticz'

I have no more idea now, why it still looking to python 3.5...

Thanks for the help

pczolee, 2021-01-26:
Well, I think the problem is this line:
 2021-01-26 10:24:58.450 Status: PluginSystem: Started, Python version '3.5.3'. 
I have no idea why using the old python, and how to tell to use the new one....

addiejanssen, 2021-01-26:
Hi, this sounds like a "sudo" problem. The domoticz process seems to be unable to find the required modules.
When you did the 'pip3 list', did you do that with your "regular" user or did you do a 'sudo pip3 list'?

pczolee, 2021-01-27:
Well, domoticz running as root, the pip3 list ran as root. Unfortunately because the old glibc I cannot update domoticz. I don't want to build from source, so I gave it up, and using from docker. With that everything looks great
Thanks

addiejanssen, 2021-01-28:
OK clear. I will close the topic then.

closed 2021-01-28 by addiejanssen