addiejanssen/domoticz-solaredge-modbustcp-plugin #31: Firmware Update & Connection Issues

jprinsen, 2022-05-18:
### Discussed in https://github.com/addiejanssen/domoticz-solaredge-modbustcp-plugin/discussions/30

<div type='discussions-op-text'>

<sup>Originally posted by **jprinsen** May 18, 2022</sup>
SolarEdge has pushed new firmware to my SE3000H Inverter causing my Modbus TCP data connection to freeze since yesterday afternoon. As a result, domoticz data has not been updated since. The current version numbers are:
DSP2: 2.100.1017
DSP1: 1.100.1146
CPU: 4.14.228

The logfile (screendump attached) states:
```
- Pushing 'onHeartbeatCallback' on to the queue
- Retrying to communicate with inverter after: [date/time]
- Processing 'onHeartbeatCallback' message
- Acquiring lock for 'onHeartbeatCallback'
- Calling message handler 'onHeartbeat' on 'module' type object
- onHeartbeat
- Retrying to comunicate with inverter after: [date/time]
- Releasing to lock for 'onHeartbeatCallback'; 
- followed by a repetitive string or acquiring/releasing lock for 'SolarEdge'...
```
[image]

I verified the system settings; ModBusTCP is still enabled; also the port number has not changed.
Still running the same (and most recent) pymodbus version (2.5.3) so we can exclude that item as well.

Anybody with some useful insights in what is causing this and how to re-establish my valuable SolarEdge data connection?
Many thanks in anticipation.</div>

jprinsen, 2022-05-18:
It appears that CPU version 4.14.XXX no longer allows ModBus TCP connections over WiFi. Is not mentioned in the changelog since 'ModbusTCP via WiFi' was never formally considered to be a 'feature' (hear-say). Will look into possibilities to enable ModBusTCP again via LAN; since my SE3000 only has WiFi I would welcome any suggestion by any of you.

addiejanssen, 2022-06-20:
This is not an issue with the software. Please discuss this the "Discussions" section.

closed 2022-06-20 by addiejanssen