addiejanssen/domoticz-solaredge-modbustcp-plugin #47: Using uninitialized self.inverter in 2.0.1

addiejanssen, 2022-10-24:
### Discussed in https://github.com/addiejanssen/domoticz-solaredge-modbustcp-plugin/discussions/46

<div type='discussions-op-text'>

<sup>Originally posted by **JohnBons** October 24, 2022</sup>
Running into an issue on my raspberry domoticz environment:
```
2022-10-24 18:13:51.079 Error: SolarEdgeMod: ----> Line 582 in '/home/bonsj/domoticz/plugins/domoticz-solaredge-modbustcp-plugin/plugin.py'
2022-10-24 18:13:51.079 Error: SolarEdgeMod: ----> Line 200 in '/home/bonsj/domoticz/plugins/domoticz-solaredge-modbustcp-plugin/plugin.py'
2022-10-24 18:14:11.078 Error: SolarEdgeMod: 'onHeartbeat' failed 'AttributeError':''NoneType' object has no attribute 'connected''.
2022-10-24 18:14:11.078 Error: SolarEdgeMod: Exception traceback:
2022-10-24 18:14:11.078 Error: SolarEdgeMod: ----> Line 582 in '/home/bonsj/domoticz/plugins/domoticz-solaredge-modbustcp-plugin/plugin.py'
2022-10-24 18:14:11.078 Error: SolarEdgeMod: ----> Line 200 in '/home/bonsj/domoticz/plugins/domoticz-solaredge-modbustcp-plugin/plugin.py'
```
Modbus itself is working (Downloaded repo and used example.py for testing
```
@raspberrypi:/solaredge_modbus $ ./example.py 192.168.2.7 1502 --json |grep device
    "c_deviceaddress": 1,
```

Environment setup:
Version: 2022.1
Build Hash: c9526851b
Compile Date: 2022-01-31 09:34:32
dzVents Version: 3.1.8
Python Version: 3.9.2 (default, Mar 12 2021, 04:06:34) [GCC 10.2.1 20210110]

pip list 
solaredge-modbus    0.7.2 (updated a few days ago)

can someone help me out?

</div>

addiejanssen, 2022-10-29:
Released 2.0.2 which should fix this issue.
Will keep it open till we get feedback from JohnBons.

JohnBons, 2022-11-02:
Fixed! Plugin is working.

addiejanssen, 2022-11-03:
Fixed by 2.0.2

closed 2022-11-03 by addiejanssen