addiejanssen/domoticz-solaredge-modbustcp-plugin #54: Spikes in data

FilipDem, 2022-12-20:
Hi,
I am running now since a while the plugin! Thanks for this! Great job!
I bought the house since a couple of months and started since then with the Domoticz plugin.

Unfortunately from time to time (once/month?) I got a spike that is messing up the charts (see image).
At a certain moment, I receive a big negative value, being the value of the total generated energy since the installation became active (but then negative)... Then a while nothing and then the same value, but with a positive value...

Anybody reported already such a problem and is there a solution/workaround?

[image]

janreimen, 2022-12-21:
Hello,

I encountered the same issue on 18-20 december, but not with the domoticz-solaredge-modbustcp-plugin 2.0.2, but with the official WebAPI. 

What I saw in the exportable .json / .xml's of the SolarEdge site is that there was wrong data at the source ... 

However, I can't find an explanation about your finding with the modbus plugin ... and I did not encounter the issue till now

janreimen, 2022-12-21:
> Hi, I am running now since a while the plugin! Thanks for this! Great job! I bought the house since a couple of months and started since then with the Domoticz plugin.
>
> Unfortunately from time to [...]

I might have an origin : @FilipDem do you use more than one inverter on your installation ?
- as I have a 3phase SE16K (being slave) connected to the solar panels, and between him and my Solar P1 Counter is an additional SE6000H 1phase (Master). At the moment I encountered the issue again 5 mins ago, the SolarEdge Data presented as follows : 
Sum kWh of Slave = y, last updated 13h38
Sum kWh of Master = x, last updated 13h51
normally, the Sum of both should be the values of x+y, but in the scenario you describe and I encouter, the sum is y !

This could explain the negative drop followed by the regain of all produced energy (in my case).

Bypassable via dzVents script replacing a native Device ... :)

FilipDem, 2022-12-22:
@janreimen : No I only have one inverter on my installation...
Any idea how we could investigate post-problem?

janreimen, 2022-12-25:
@FilipDem  : I don't have the exact official answer from SolarEdge, but when I look at my realtime data / logs in my installation, I tend to agree with your assumption. 

Until now (I'm using domoticz since 2 years), the issue never happened for me at the end-of-day, so the wrong data was only on the Meter/Multimeter Table and not in the Meter_Calendar / Multimeter_Calendar table. Even though, the Web Interface allows in this table to delete on unique point (i.ex. on a Temperature Node, you can delete ex-post a wrong value like 0° on the Modbus Temperature of the Inverter ... ). Alternatively, you pass as DBA to the domoticz.db file with a CLI like sqlite3 or a Frontend like sqlitebrowser ...