addiejanssen/domoticz-solaredge-modbustcp-plugin #73: increase version of solaredge_modbus

JanJaapKo, 2024-02-07:
### Discussed in https://github.com/addiejanssen/domoticz-solaredge-modbustcp-plugin/discussions/72

<div type='discussions-op-text'>

<sup>Originally posted by **JanJaapKo** February  7, 2024</sup>
solaredge_modbus version 0.7.0 doesn't work anymore (at least not on new install), current version is 0.8.0.</div>

infoedkok, 2025-01-06:
Be aware that solaredge_modbus 0.8.0 also requires pymodbus==3.5.1. Otherwise, you will get errors like 
"SolarEdgeMODBUS: ModuleNotFoundError: No module named 'pymodbus.register_read_message" from pymodbus.register_read_message import ReadHoldingRegistersResponse

JanJaapKo, 2025-02-15:
Sorry for late reply, but the committed change does not lead to any issues (perhaps/probably handled in the dependencies of solaredge_modbus )

MadPatrick, 2025-05-05:
Hi

When using this plugin with the latest version i get the following error
pyserial-asyncio-0.6 
solaredge_modbus-0.8.0
pymodbus-3.9.2

`2025-05-05 17:01:21.110 Error: SolarEdge TCP: (SolarEdge_ModbusTCP) failed to load 'plugin.py', Python Path used was '/home/patrick/domoticz/plugins/domoticz-solaredge-modbustcp-plugin/:/usr/lib/python311.zip:/usr/lib/python3.11:/usr/lib/python3.11/lib-dynload:/usr/local/lib/python3.11/dist-packages:/usr/lib/python3/dist-packages:/usr/lib/python3.11/dist-packages'.
2025-05-05 17:01:21.111 Error: SolarEdge TCP: Traceback (most recent call last):
2025-05-05 17:01:21.111 Error: SolarEdge TCP: File "/home/patrick/domoticz/plugins/domoticz-solaredge-modbustcp-plugin/plugin.py", line 54, in <module>
2025-05-05 17:01:21.111 Error: SolarEdge TCP: import solaredge_modbus
2025-05-05 17:01:21.111 Error: SolarEdge TCP: File "/usr/local/lib/python3.11/dist-packages/solaredge_modbus/__init__.py", line 9, in <module>
2025-05-05 17:01:21.111 Error: SolarEdge TCP: from pymodbus.register_read_message import ReadHoldingRegistersResponse
2025-05-05 17:01:21.111 Error: SolarEdge TCP: ModuleNotFoundError: No module named 'pymodbus.register_read_message'`

MadPatrick, 2025-05-31:
This works now with the latest verison

pymodbus==3.6.9
solaredge_modbus==0.8.0

addiejanssen, 2025-12-23:
Updated the requirements.txt file to use the latest supported version.

closed 2025-12-23 by addiejanssen